ros2/common_interfaces #3: Migrate sensor_msgs utilities to ROS 2

wjwwood, 2015-06-17:
ROS 1's `sensor_msgs` contains C++ headers and Python modules that provide some utilities for the generated message code data types, e.g. filling an image message or iterating over a point cloud.

I avoided bring that with the current migration effort. We should decide if we want to host those utilities elsewhere of if we want them in the `sensor_interfaces` package, similar to how it is laid out in ROS 1. It seems like it would be best if those things were in a `sensor_interfaces_utilities` package instead, but maybe there is a good reason for the current colocation in ROS 1.

TODO: update these entries in the migrated messages once this is done:

- https://github.com/ros2/common_interfaces/blob/7bcca4969e891911d6ba99dd2093c75132279c5b/sensor_interfaces/msg/CameraInfo.msg#L14
- https://github.com/ros2/common_interfaces/blob/7bcca4969e891911d6ba99dd2093c75132279c5b/sensor_interfaces/msg/CameraInfo.msg#L58
- https://github.com/ros2/common_interfaces/blob/7bcca4969e891911d6ba99dd2093c75132279c5b/sensor_interfaces/msg/Image.msg#L22

Also look into avoiding things like valid encoding values being captured in C++ headers and instead capture them in constants in the messages. Again, there might be a good reason for this, but it smells to me.

clalancette, 2017-05-11:
Just FYI, I have done parts of this as needed for the cartographer integration effort.  I think we ended up deciding to just keep these utilities as part of the same sensor_msgs package for ease of use/porting from ROS1.  That being said, I haven't done all of the utilities, just the ones I've needed (point cloud iterators and distortion models), so we should leave this open until we decide what to do about the rest.

mikaelarguedas, 2017-05-11:
related PRs: pointcloud2 iterators: #34 and distortion models #31.

Also if we decide to lint these migrated files we should also make sure they comply with our developer guide (doesn't seem to be the case AFAICT)

sagniknitr, 2018-07-17:
Any updates on regarding migrating these utilities ? We need fill_image.h for our software stack for ROS 2.0.

mikaelarguedas, 2018-07-30:
Thanks @sagniknitr for your interest in this.

We didn't have the need for the rest of the utilities so far.
We will happily review any PR that port additional utilities.

jacobperron, 2019-09-05:
It appears all of the utilities have been migrated. I've opened a PR updating the references mentioned in the description (#72). Regarding the reference to CameraInfo details (http://www.ros.org/wiki/image_pipeline/CameraInfo), it's currently referencing the original documentation under `image_pipeline`. Perhaps we should consider moving that documentation to this repository to live closer to the message definition. In any case, I think it is out of scope for this issue.

closed 2019-09-05 by jacobperron