ros2/common_interfaces #4: merge builtin_interfaces with std_interfaces

codebot, 2015-07-01:
I propose that we merge builtin_interfaces with std_interfaces, primarily because I don't see why Time and Duration are "different enough" to justify their own interface package.

People will be typing those identifiers enough that having them in an "obvious" place next to other "obvious" lowest-level types seems valuable.

dirk-thomas, 2015-07-01:
The types `Duration` and `Type` are higher level then the primitive types but lower level then all user defined types. The `builtin` types are mapped to DDS native types and must therefore be treated specially in the type support packages / rmw implementation. Therefore I think the separation make sense from a technical point of view.

wjwwood, 2015-07-01:
+1

Technical note: we'll have to update our special handling of these types within opensplice message generation, but that should easy enough. 

One other draw back is that these messages are the successors to the time and duration built-in types from ROS 1, but there were also std_msgs/Time and Duration which are no longer there. So that might cause some confusion.

codebot, 2015-07-01:
we can treat them specially regardless of what package they are in; the block of code looking for them just needs to look at {package name, message name} instead of just {package name}.

std_interfaces/msg also has ColorRGBA, which is quite similar: it's a canonical way of representing something quite fundamental. I think we should be most concerned about the end-user's perspective, and understanding why something is in builtin_interfaces vs std_interfaces is not obvious unless you understand the details of the implementations at a level beyond what I think is fair to most end users.

esteve, 2015-07-01:
+1 to merging them, I agree with @codebot about the user experience.

dirk-thomas, 2015-07-01:
For most messages in std_interface we highly recommend users *not* to use them. Therefore I am not sure if that is the right package for Duration and Time.

Technical note: when the builtin messages are being moved the `ros1_bridge` also needs to be updated.

wjwwood, 2015-07-01:
I agree that many of them are just for convenience to use directly. In fact really only `Empty` could be considered a valid message to use directly as the type for a topic. But for indirect use, i.e. use within another message definition, there are some notable exceptions like Header, ColorRGBA, and the MultiArray* messages. I think Time and Duration fall in the latter category, that is to say you probably shouldn't use them directly as the type for a topic, but you will often use them in other message definitions.

codebot, 2015-07-01:
I agree with @wjwwood that std_interfaces has a mixture of "useful" things (Empty, Header, ColorRGBA, MutliArray*) and "historical" things that are singletons of all the primitive types.

If we want to discourage people from using the "singleton" std_interfaces/msg types directly, I think we should do that explicitly by making another interfaces package called "singleton_msgs" or "old_std_msgs" or "woah_there" or "ahhhhh_msgs" or something more obvious. I think the way to discourage use of the pre-baked "singleton" message types would be to (magically, somehow, I don't know how exactly) have the creation of new messages be much easier; maybe automatic, etc. etc., so you wouldn't have to touch CMake when you are just hacking stuff together in Python. 

But anyway, I think that removing cruft from std_interfaces is a separate issue from adding Time/Duration to std_interfaces. The "std" and "builtin" keywords are just so close semantically (to me, at least) that I could see it causing confusion. I recognize this will cause some of our code thus far to require updating, but that problem will just get worse over time.

gerkey, 2015-07-02:
+1 to merging into `std_interfaces`.

As an aside: while it's not *best* practice to use semantically meaningless topic types like time, color, integer, etc., I wouldn't go so far as saying that people *shouldn't* use them in that way.  Quick prototyping has always been a strength of ROS, and being able to easily throw around messages containing primitive data types helps to make that happen.

mikaelarguedas, 2017-08-08:
builtin_interfaces have been merged in this repo, closing this

closed 2017-08-08 by mikaelarguedas