ros2/common_interfaces #11: builtin_interfaces doesn't generate python msgs in arm

ahcorde, 2016-02-18:
Hi,

I'm trying to compile ROS2 Alpha4 from it sources in `armhf`. Before trying to compile in `armhf` I have compiled in my `x86` and everything works fine.

The issue is in the package `builtin_interfaces` during the installation process. It looks like the python msgs are not generated.

>  CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:155 (message):
>   ament_cmake_symlink_install_files() can't find
>   '/home/erle/ros2_ws/build/builtin_interfaces/rosidl_generator_py/builtin_interfaces/msg/_duration.py'
> Call Stack (most recent call first):
>   ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:344 (ament_cmake_symlink_install_files)
>   cmake_install.cmake:36 (include)

ahcorde, 2016-02-18:
@vmayoral can you reproduce it?

vmayoral, 2016-02-18:
Got a similar result https://gist.github.com/vmayoral/fa3c129034a137194fca. `builtin_interfaces ` fails.

jacquelinekay, 2016-02-18:
We have not tested the Python message generation/client library on armhf yet, and we haven't set up a CI job for ARM. Thanks for catching this.

If you don't need Python, one workaround would be to add an empty AMENT_IGNORE file to rosidl_generator_py and rclpy.

In the meantime we'll investigate the issue.

esteve, 2016-02-18:
@ahcorde @vmayoral can you check if this branch (https://github.com/ros2/rosidl/tree/generate-py-before-install) fixes the problem? Thanks!

ahcorde, 2016-02-18:
@esteve, it fails at the same point.

vmayoral, 2016-02-18:
Thanks @esteve and @jacquelinekay for the the quick replies.
The following did it for me:
```
diff --git a/rosidl_default_generators/CMakeLists.txt b/rosidl_default_generators/CMakeLists.txt
index 6e7ae72..a8be083 100644
--- a/rosidl_default_generators/CMakeLists.txt
+++ b/rosidl_default_generators/CMakeLists.txt
@@ -8,7 +8,7 @@ ament_export_dependencies(rosidl_cmake)
 
 ament_export_dependencies(rosidl_generator_c)
 ament_export_dependencies(rosidl_generator_cpp)
-ament_export_dependencies(rosidl_generator_py)
+#ament_export_dependencies(rosidl_generator_py)
 
 if(AMENT_ENABLE_TESTING)
   find_package(ament_lint_auto REQUIRED)
```
```
diff --git a/rosidl_default_generators/package.xml b/rosidl_default_generators/package.xml
index c0923c5..3baa723 100644
--- a/rosidl_default_generators/package.xml
+++ b/rosidl_default_generators/package.xml
@@ -12,7 +12,7 @@
 
   <buildtool_export_depend>rosidl_generator_c</buildtool_export_depend>
   <buildtool_export_depend>rosidl_generator_cpp</buildtool_export_depend>
-  <buildtool_export_depend>rosidl_generator_py</buildtool_export_depend>
+  #<buildtool_export_depend>rosidl_generator_py</buildtool_export_depend>
 
   <buildtool_export_depend>rosidl_typesupport_introspection_c</buildtool_export_depend>
   <buildtool_export_depend>rosidl_typesupport_introspection_cpp</buildtool_export_depend>

```

esteve, 2016-02-18:
@ahcorde thanks. It seems from @vmayoral'ss trace that the Python step does not run at all, maybe `${_generated_msg_py_files}` is empty in https://github.com/ros2/rosidl/blob/master/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake#L174 and thus the Python generator does not run at all? @jacquelinekay what do you think?

esteve, 2016-02-18:
@ahcorde @vmayoral are you using Ubuntu Trusty on arm? Does the same error happen on x86 with the same OS? We're working on reproducing this error, but it'd be great if we had the same environment as yours.

ahcorde, 2016-02-18:
**x86** Ubuntu Trustry 14.04: works fine
**armhf** Debian 8 (Jessie) fails as describe above

jacquelinekay, 2016-02-18:
Just out of curiosity, which DDS/RTPS are you using and how did you install it?

esteve, 2016-02-18:
@ahcorde @vmayoral did you install a DDS vendor? From @vmayoral 's log, it seems that CMake only finds introspection, but not opensplice or connext, which are required for rosidl_generator_py

esteve, 2016-02-18:
Also, what does /etc/apt/sources.list.d/gazebo-latest.list look? If it has "jessie" in it, could you try changing it to "trusty" and build again? We now have armhf packages for opensplice, but only for Trusty.

esteve, 2016-02-18:
Oops, we don't have the opensplice package for ARM in our repo. You'll have to install it manually from http://www.osrfoundation.org/tmp/libopensplice64_6.4.0-3~trusty_arm64.deb

vmayoral, 2016-02-18:
@esteve,
We are compiling opensplice from its sources. [This](http://erlerobotics.com/docs/Robot_Operating_System/ROS_2/Installing_ROS_2.html) did it back in alpha2 but maybe we are missing something this time. I'll retry everything once again tomorrow. Thanks!

jacquelinekay, 2016-02-18:
builtin_interfaces compiles for me with 14.04 Ubuntu on armhf (in a container) and the Opensplice deb linked above.

Perhaps the Opensplice config file needs to be sourced.

We will work on making rosidl_generator_py fail sooner with a more helpful error mesage.

jacquelinekay, 2016-02-19:
I was able to reproduce this error message on our build farm (14.04, x86) by disabling Opensplice and Connext.

http://ci.ros2.org/job/ci_linux/1027/console

vmayoral, 2016-02-19:
Thanks for the effort @jacquelinekay. I've gone ahead and started over again from a clean Debian 8 Jessie (armhf, 32 bits) file system. Compiled opensplice again from its sources and made sure the environment variables were as described in the instructions.

With this, the previous error related to `builtin_interfaces` didn't show anymore so likely we had something wrong (opensplice related probably). Although I passed through the `builtin_interfaces` issue, I found the following hurdles:
- intra_process_demo ([error log](https://gist.github.com/vmayoral/3de928e5b0cce6917f07))
- tf2_ros ([error log](https://gist.github.com/vmayoral/c4eba09d48e3cd8aaadb))

I'd say we can close this.

jacquelinekay, 2016-02-19:
Those pthread linking errors are consistent with reports we've heard from another Debian user: ros/geometry_experimental#140 Not sure if we're going to merge that PR, however.

Thanks for checking this out! :cake:

closed 2016-02-19 by jacquelinekay