ros2/common_interfaces #27: Could not find the sensor_msgs/BatteryState?

sidd1, 2017-01-19:
I was not able to locate the sensor_msgs/BatteryState msg anywhere in the ros2 repo. However I did notice that all the other messages from sensor_msgs have been ported. Am I missing something?

mikaelarguedas, 2017-01-19:
I think the reason is that `BatteryState` has been added to `sensor_msgs` after the initial port of this package to ROS2. I don't see a reason not to port it too. Thanks for pointing it out

sidd1, 2017-01-19:
Following up, the `Header` message in `std_msgs` does not have the `seq` field. Could be the same reason that it was updated after the port of this package.

codebot, 2017-01-19:
The DDS/RTPS protocol (and many/most middlewares in general) includes a sequence number for each message for internal flow control, so the idea was the `seq` field in the ROS message header would be redundant. In the RTPS case, there is some text describing the sequence number field on page 51 of the [official spec PDF](http://www.omg.org/spec/DDSI-RTPS/2.2/PDF/), and more details of this field throughout the document. Cheers

mikaelarguedas, 2017-01-19:
Most of the high level changes between ROS1 and ROS2 are documented [here](http://design.ros2.org/articles/changes.html) if you want to have an overview without diving into the details of the spec

sidd1, 2017-01-19:
Ah, I missed that. Thanks!

tfoote, 2017-01-19:
Also note that the seq element has been deprecated since noteably before 2013. It was left for backwards compatibility. The oldest link I can find is here: http://answers.ros.org/question/55126/why-does-ros-overwrite-my-sequence-number/?answer=55132#post-id-55132

wjwwood, 2017-01-19:
See this discussion: https://groups.google.com/forum/#!topic/ros-sig-ng-ros/_UFMxG5TG9s

closed 2017-03-02 by mikaelarguedas