ros2/common_interfaces #32: linters removed for sensor_msgs

dhood, 2017-04-25:
There has been a commit to master to disable all linters on the sensor_msgs package (a0234f2254221b9927ede3fe06dd6abc2d95559a).

There may have been some offline discussion about this already, but if not then @clalancette could you please re-enable linters on this package and only disable the copyright? I have seen that you've opened https://github.com/ament/ament_lint/issues/72 in the meantime :+1:

clalancette, 2017-04-25:
No, this only disables the auto linter, which was only used to lint include/sensor_msgs/distortion_models.hpp.  The linting for the message generation still happens via: 

)
rosidl_generate_interfaces(${PROJECT_NAME}
  ${msg_files}
  ${srv_files}
  DEPENDENCIES builtin_interfaces geometry_msgs std_msgs
  ADD_LINTER_TESTS
)

mikaelarguedas, 2017-04-25:
I think @dhood meant disabling only the copyright test and not all linters on include/sensor_msgs/distortion_models.hpp. But you're right the generated messages are still linted.

The linters being disabled for code in this package this issue should stay open.

clalancette, 2017-04-25:
Oh, I see.  Right, OK, let's do that :).

closed 2017-05-02 by clalancette