ros2/common_interfaces #44: Marker.msg DELETE field conflicts with windows SDK 8.1

dejanpan, 2017-10-27:
## Bug report
It looks like I've found one more conflict with Windows SDK 8.1.
Windows SDK defines DELETE macro in winnt.h.
src\ros2\common_interfaces\visualization_msgs\msg\Marker.msg uses
uint8 DELETE=2

**Required Info:**

- Operating System:
  Win 8.1
- Installation type:
source
- Version or commit hash:
beta3-release
- DDS implementation:
RTI Connext
- Client library (if applicable):
rmw_connext_cpp

#### Steps to reproduce issue
`ament build`

#### Expected behavior
compile sucesss

#### Actual behavior
compile fails

#### Additional information
@serge.nikulin

serge-nikulin, 2017-10-27:
@dejanpan, I'm @serge-nikulin here.

wjwwood, 2018-06-15:
Another instance where this came up: https://github.com/ros2/rviz/pull/296/files#diff-04fd38f59f6b88e57db1feea94b3fa4cR44

serge-nikulin, 2018-06-15:
Are you going to address it somehow?

wjwwood, 2018-06-15:
Well in `rviz` we're using the recommended workaround which is to `undef DELETE` locally.

There's an effort underway to address the problem more generally, please ask there.

https://github.com/ros2/design/pull/172
https://github.com/ros2/rosidl/issues/118

serge-nikulin, 2018-06-15:
nah, I'm good unless we support Windows formally.