ros2/common_interfaces #48: Add Point2D and Vector2

calvertdw, 2018-02-27:
Is there a reason these aren't included?

More importantly, is there a reason why we shouldn't make these for ourselves and start using them in our API?

Is there an official place that they exist elsewhere?

dirk-thomas, 2018-02-27:
This repository is a fork of the existing ROS 1 repository [ros/common_msgs](https://github.com/ros/common_msgs). We didn't add / change / remove any in this port. So if these types are not defined for ROS 1 they are also not for ROS 2. Please consider filling a ticket "upstream" instead.

dirk-thomas, 2018-02-27:
@calvertdw If your question has been answered and there is nothing further to be done on this ticket please close it. Thanks.

wjwwood, 2018-02-28:
If you file a question/pr upstream, it would be useful to link it here too.

As to the actual question, I think we prefer to use Point3D when representing a 2D point so that tools are more reusable. For instance, rviz might not be able to render a 2D point, but it can render a 3D point which always has z=0. So slightly favoring reusability of tools over the extra bandwidth on the wire which is wasted representing a 2D point with a 3D message.

tfoote, 2018-02-28:
This was reasked at: https://github.com/ros/common_msgs/issues/121 Closing as a duplicate.

closed 2018-02-28 by tfoote