ros2/common_interfaces #49: Add message with JointStates and odometry

jespersmith, 2018-03-26:
----
## Feature request

Add a message with JointStates and the root translation/orientation with respect to world.

#### Feature description

To get an atomic representation of the robot state, without the delay of waiting for a TF2 message, it would be nice to have a single message that includes both the joint states as well as the root joint transform to world.

#### Implementation considerations

Several solutions come to mind

- A merger between MultiDoFJointState and JointState, were you can freely specify a combination of JointStates with 1 or 6 DoF's.

- Adding a single Transform to JointState. However, this will not be useful for robotic systems with multiple free floating root joints that are controlled by a single monolithic controller.

A solution would be to use MultiDoFJointState for 1 degree of freedom joints, however it can be tricky to extract the joint angles from quaternions and there is a large overhead in data usage.

tfoote, 2018-03-27:
I'm not sure that this makes sense. What are you defining as the "world" and how do you know that value?

clalancette, 2022-04-29:
No response in many years to the question, so I'm going to close this out.  Feel free to reopen if you still want to pursue this.

closed 2022-04-29 by clalancette