ros2/common_interfaces #66: Port point_cloud_conversion and point_field_conversion

mjcarroll, 2019-01-11:
## Feature request

Related to: #3, created via ros2/ros2#641

Port `sensor_msgs/point_cloud_conversion.h` and `sensor_msgs/point_field_conversion.h`

#### Feature description

`sensor_msgs` should contain utility methods for working with PointClouds and PointFields as it did in ROS1

#### Implementation considerations

Generally follow the process used in #58, by copying the ROS1 equivalent file and updating accordingly.

vandanamandlik, 2019-01-14:
@mjcarroll , I think following links will be helpful to you, in that I have copied point_cloud_conversion.h and point_field_conversion.h in laser_assembler package and ported to ROS2.
[https://github.com/vandanamandlik/laser_assembler/blob/ros2-crystal/include/laser_assembler/point_cloud_conversion.hpp](url)
[https://github.com/vandanamandlik/laser_assembler/blob/ros2-crystal/include/laser_assembler/point_field_conversion.hpp](url)

mjcarroll, 2019-02-19:
Closed by #67

closed 2019-02-19 by mjcarroll