ros2/common_interfaces #76: Add YUYV image encoding

sgvandijk, 2019-11-09:
----
## Feature request

#### Feature description
Most popular USB cameras that are used in mobile/hobby robots, such as the Logitech C9xx cameras, provide images in YUV 4:2:2 format in YUYV order.

There is a `YUV422` encoding defined in `image_encodings.hpp` but that is defined to be in UYVY order.

With an encoding constant defined for YUYV order we can then go on to handle that format in e.g. `cv_bridge` to convert it to other formats so there are no custom solutions needed like [done in the ROS 1 `usb_cam` driver](https://github.com/ros-drivers/usb_cam/blob/develop/src/usb_cam.cpp#L327).

#### Implementation considerations
The implementation will be almost the same as for `YUV422`. The main consideration I think is the naming. Looking at https://www.fourcc.org/yuv.php the canonical name for this format seems to be `YUY2`, with `YUYV` being a synonym. OpenCV indicates the same in its [conversion codes (e.g. `COLOR_YUV2RGB_YUYV = COLOR_YUV2RGB_YUY2`)](https://docs.opencv.org/master/d8/d01/group__imgproc__color__conversions.html).

With that I'd propose either:
1. `YUY2`
2. `YUV422_YUY2` <-- seems best to me
3. `YUV422_YUYV`

I'd be happy to make a PR for this if this is accepted and a name is decided, and I then plan to add handling of it in `cv_bridge` as I'd like to use that in our [V4L2 camera driver](https://gitlab.com/boldhearts/ros2_v4l2_camera).

clalancette, 2019-11-14:
I'd go for either 1 or 2.  I have a slight preference for 1 (as it is less verbose), but I defer to your judgement on which is clearer.  Please go ahead and open a PR for it; when you do, please be sure to include a link to exactly which pixel format you are talking about.  Thanks!

mjcarroll, 2019-11-14:
I'm good with either 1 or 2 with a preference towards 2.  Looking forward to the PR!

clalancette, 2019-11-25:
Fixed by #78, so closing.

closed 2019-11-25 by clalancette