ros2/common_interfaces #95: Improve documentation of nav_msgs/GridCells

tfoote, 2020-04-07:
This is a follwup to the [pre Foxy Message API review](https://github.com/ros2/common_interfaces/pull/86) 

Are the points in cells the center point? It could use some documentation?

tfoote, 2020-05-18:
@SteveMacenski @dlu could you help fill in this documentation for clarity?

SteveMacenski, 2020-05-18:
On GridCells: I can't find anyone that uses that other than grid_maps. It looks like from their code they use it with RVIZ so I think we should be able to derive any information about it from that implementation. 

On MapMetaData: The message has the comment in it

```
# The origin of the map [m, m, rad].  This is the real-world pose of the
# cell (0,0) in the map.
```

which I think makes that clear. [This diagram](https://answers.ros.org/question/205521/robot-coordinates-in-map/) gives a good overview of the coordinate system.

DLu, 2020-05-18:
> GridCells - Are the points in cells the center point? It could use some documentation?
Based on [this](https://github.com/ros-planning/navigation/blob/groovy-devel/costmap_2d/src/costmap_2d_publisher.cpp#L91) the coordinates are the same as `mapToWorld` which is the [center](https://github.com/ros-planning/navigation/blob/groovy-devel/costmap_2d/src/costmap_2d.cpp#L546). In `rviz` the grid cells are displayed as a [point cloud](https://github.com/ros-visualization/rviz/blob/171eb3cfd1ec69cb6ff72da3ca41a1a4caa74eef/src/rviz/default_plugin/grid_cells_display.cpp#L246) which confirms that the coordinates are the center. 

> Is origin the center point of the (0,0) cell? 
No it is the bottom left corner of the (0,0) cell

> Which direction do cell coordinates increase? 
If saying bottom left above wasn't a clue, x to the right, y up

> Is (0,0) x,y in a right handed coordinate system such that (0,0) is the bottom right cell, or is it row,col such that (0,0) is the top left cell?
I think a better way to describe it is that it follows the same coordinates as the world if the resolution is 1m. 

[I have additional documentation here](https://github.com/locusrobotics/robot_navigation/tree/master/nav_grid). While not explicitly for the OccupancyGrid, it does use largely the same math. 

Back in the day, [`costmap_2d` used to publish GridCells](https://github.com/ros-planning/navigation/blob/groovy-devel/costmap_2d/src/costmap_2d_publisher.cpp#L44) but that functionality has been removed.

PR #117 (2020-05-19): Add comments for nav_msgs defintions
Fixes #95
closed 2020-05-19 by tfoote