ros2/common_interfaces #104: consider extending NavSatFix with an optional orientation

tfoote, 2020-04-08:
This is a follwup to the [pre Foxy Message API review](https://github.com/ros2/common_interfaces/pull/86) 

Consider a extending the message with an optional orientation. Or consider a new extended message with orientation added. This message is specifically generic to all GNSS systems versus the GPS one has GPS specific data so is not suitable to be replaced.

Ryanf55, 2023-04-04:
Could this be targeted for ROS2 Iron?

SteveMacenski, 2023-09-13:
I suggested potentially including a `NavSatFixWithOrientation` kind of message that combines them, in case we want to add a new message instead of modifying an old one. A NavSatFix by itself usually doesn't have orientation information for basic GPS systems. However, most new (non-cheapo) GNS systems in robotics have IMUs and such for better fusion, so I think that's worthy of consideration into `sensor_msgs` to include orientation to modernize.

@Ryanf55 can you bring this up in the GPS thread on discourse? It would be good to have everything in a single spot to compile a summary from once thoughts are collected

Ryanf55, 2023-09-14:
> I suggested potentially including a `NavSatFixWithOrientation` kind of message that combines them, in case we want to add a new message instead of modifying an old one. A NavSatFix by itself usually [...]

Reference: https://discourse.ros.org/t/future-of-ros-2-gps-support/33297/20