ros2/common_interfaces #133: [sensor_msgs/PointCloud2Iterator] Iterator code does not get vectorized when used in loops

YoshuaNava, 2020-10-19:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- Installation type:
  -  Debians
- Version or commit hash:
  -  
- DDS implementation:
  - 
- Client library (if applicable):
  -  ros 1

#### Steps to reproduce issue

```
        sensor_msgs::PointCloud2ConstIterator<T> iterX(rosMsg, fieldNames[0]);
        sensor_msgs::PointCloud2ConstIterator<T> iterY(rosMsg, fieldNames[1]);
        sensor_msgs::PointCloud2ConstIterator<T> iterZ(rosMsg, fieldNames[2]);

        for (size_t i = 0; i < pointCount; ++i, ++iterX, ++iterY, ++iterZ)
        {
            view(0, i) = *iterX;
            view(1, i) = *iterY;
            view(2, i) = *iterZ;
        }
```

In the above code snipped, view is a block of an Eigen matrix, pointCloudMsg is a ROS message of type sensor_msgs/PointCloud2. I'm filling the data from the buffer into the matrix.

#### Expected behavior

I would expect the above loop to be auto-vectorized by my compiler (gcc 9.3.0). However, it seems to be running in scalar mode. I'm building in RelWithDebInfo with O3.

The iterator is a really cool component that is shipped with sensor_msgs, but I might be using it wrong, or there might be some element in the code that makes it difficult for the compiler to figure out how to simplify the above construct and vectorize it.

#### Actual behavior

The loop presented above runs in scalar mode, it's fast. But I would still like to still try to vectorize it, because I have to fulfill a resource limit.

#### Additional information

Only some questions:

1. Am I using the iterator in the right way?
2. Why are there so many casts in the operators from the iterator?
3. Am I traversing the msg buffer in the most efficient way? Should I re-think the loop?

YoshuaNava, 2020-10-19:
I just noticed that I opened this issue on the ros2 repo by mistake. I will keep it here but reference it in the original repo.

YoshuaNava, 2020-10-21:
Hi @vrabaud,
As the author of the PointCloud2Iterator I would like to kindly ask you for your input on this issue. It is a really cool iterator implementation, and I have 1-2 days which I could spend looking into any possible improvement to speed it up.

Thank you in advance. 

PS: Sorry if this is not the right way/repo to contact you, but I also wanted to prevent creating a duplicate issue in the ROS1 repo.

clalancette, 2020-11-05:
This repository only deals with ROS 2, so I'm going to close this ticket out.

If you'd like to follow-up with ROS 1, feel free to open up a ticket at https://github.com/ros/common_msgs

closed 2020-11-05 by clalancette

YoshuaNava, 2020-11-05:
@clalancette

Sorry to bother. The iterator is part of the code from this repo, and I guess it's used in other ROS2 packages as a consequence.