ros2/common_interfaces #142: add sensor_msgs_py

flynneva, 2021-03-10:
## Feature request
add sensor_msgs_py to Foxy

#### Feature description
Foxy will be the default distro for awhile now and the `sensor_msgs_py` has already been merged into the master branch for this repo. I was just wondering if it would be possible to also release it for the foxy branch as well.

#### Implementation considerations
should just be a PR/merge I think. ive tested it and it compiles just fine

hidmic, 2021-03-15:
It is fairly uncommon for a feature (let alone an entire package) to be backported to a previous distro. Though I wouldn't expect this particular addition to compromise downstream code in any way. @jacobperron what do you think about a Foxy backport?

flynneva, 2021-03-15:
@hidmic, completely agree - however for the most part this `sensor_msgs_py` package is required if a ROS2 user would like to work with Pointcloud2 messages in python i would say.

for that reason i think its pretty critical to get it released for Foxy since most people will be using Foxy probably until the end of the year if not into 2022 I would say.

you can follow the discussion of this package in the PR that added it to ROS2: https://github.com/ros2/common_interfaces/pull/128

there is some debate where it should reside...but that i think is a problem that could be addressed after its available "somewhere".

flynneva, 2021-03-15:
if you'd like to see how this package is used, you can check out the example I added too that uses it: https://github.com/ros2/examples/pull/276

jacobperron, 2021-03-16:
Sounds okay to me. I don't expect any disruption related to releasing the new packages into Foxy.

@flynneva Would you like to open a PR to the `foxy` branch?

flynneva, 2021-03-16:
opened here #143

jacobperron, 2021-03-18:
Done in #143

closed 2021-03-18 by jacobperron