ros2/common_interfaces #144: package builds default to underlayed sensor_msgs instead of local workspace version

mrawding, 2021-03-18:
## Bug report

**Required Info:**

- Operating System:
  - 
Ubuntu 20.04LTS
- Installation type:
  - 
  - binaries
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

I have pulled ros2/common_interfaces into my workspace and edited sensor_msgs to adapt  it to NV21 and NV12 pixel formats. However, when I go to build cv_bridge it uses the ROS2 installed version of sensor_msgs vs the adapted one in the workspace. This causes failure. However, in ROS1 this would not happen with catkin, it would default to the package in the workspace. How can I change colcon to behave like it did in ROS1?

i.e., when I build cv_bridge it should use the latest build of sensor_msgs in my work space. Instead of defaulting to the underlayed ROS2 binary sensor_msgs

I would submit a pull-request because I know it works in ROS1 but there are other packages like image_pipeline and cv_bridge that would also need pull request so that the rest of ROS is functional.

hidmic, 2021-03-26:
@mrawding would you mind being more specific about workspace layout and `colcon` usage? I'd bet this is user error (in which case I'd strongly suggest you to go ask a question to answers.ros.org), but maybe just maybe it's not.

mrawding, 2021-03-26:
User error! You can close this issue.

closed 2021-03-26 by clalancette