ros2/common_interfaces #157: Adding msg support to true zero copy.

ZhenshengLee, 2021-08-06:
## Feature request

#### Feature description

zero copy is a shm based transport layer of ros2 that can reduce cpu usage and latency, especially for large size of data like pointcloud or image.

the mainstream of zero copy of ros2 is cyclonedds+iceoryx, see [apex's ros2_shm_demo](https://github.com/ApexAI/ros2_shm_demo)

I have created a repo [ros2_shm_msgs](https://github.com/ZhenshengLee/ros2_shm_msgs) and give some pointcloud msg definitions and an early demo.

The problem is there will be a lot of definitons according to the size of pointcloud.

see [2021-08-05-Eclipse-iceoryx-developer-meetup](https://github.com/eclipse-iceoryx/iceoryx/wiki/2021-08-05-Eclipse-iceoryx-developer-meetup) for details.

So is there a method to define a parameterization of msg definition? 

Or is there any way to workaroud?

Thanks.

tfoote, 2021-08-06:
As discussed in the notes linked, creating N arbitrarily sized messages will break compatibility with pretty much all systems. 

At the highest level we have parameters for the size of the point clouds at runtime. What you're looking for that can be compiled in fixed size limits are not supported. There are possibilities to have bounded size arrays in messages. But depending on the application the appropriate upper bound is different. To support pre-allocation a discussion will need to be had about how to build and extend the message generators etc. Just adding new arrays of new message definitions isn't a workable solution. This needs to be a higher level discussion about extending [rosidl](https://github.com/ros2/rosidl) and the message definitions. Documentation is [here](https://docs.ros.org/en/foxy/Concepts/About-ROS-Interfaces.html) Somewhere like the [realtime working group ](https://docs.ros.org/en/foxy/Governance.html#working-groups-wgs) or on [NG ROS](https://discourse.ros.org/c/ng-ros/25) category with a proposal for how to extend support for fixed size messages might make sense. But overall this is a hard problem that's likely going to require structural changes to deal with properly. This repository isn't the right place for the discussion so I'm going to close it here.

closed 2021-08-06 by tfoote

ZhenshengLee, 2021-08-08:
> As discussed in the notes linked, creating N arbitrarily sized messages will break compatibility with pretty much all systems.
>
>
>
> At the highest level we have parameters for the size of the point [...]

Thank you very much!