ros2/common_interfaces #174: Request to add std_srvs

SeungRyeol, 2022-02-25:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description
I wish there was a set_int.srv or set_float.srv in std_srvs.
Users can make a custom service, but I think it would be very useful to use frequently, such as std_srvs/srv/set_bool.
Can you add the int or float type in std_srvs?

#### Implementation considerations

- std_srvs/srv/set_int
```bash
typedef struct std_srvs__srv__SetInt_Request
{
  int data;
} std_srvs__srv__SetInt8_Request;
```

clalancette, 2022-02-25:
We explicitly do not have these services, because ROS 2 is meant to be a strongly-typed system that expresses semantics.  Having a `SetInt` service is strongly typed, but expresses no semantics whatsoever; you don't know if that is controlling an LED on an arduino, or changing how your A* planner operates.

So for that reason, we are very unlikely to do this.  I'll leave this open for now for more discussion, but there would have to be a very convincing reason for us to change course on this.

gbiggs, 2022-02-27:
I agree with @clalancette. Having services without semantics is dangerous and reduces the maintainability and usability of nodes that use them.