ros2/common_interfaces #180: sensor_msgs/Range lacks variance field

ejalaa12, 2022-03-16:
Hi, 
As previously asked by @okalachev in the ros1 counterpart: https://github.com/ros/common_msgs/issues/142

> All the sensor related messages have variance/covariance field: FluidPressure, Illuminance, Imu, MagneticField, NavSatFix, RelativeHumidity, Temperature; while Range lacks it.
> 
> Though rangefinders' datasheets often provide such information, and it can be quite useful for the consumers of rangefinder drivers.

Is it something that you would consider for ros2 as well ?

Thank you.

gbiggs, 2022-03-16:
Yes, if a PR were provided I think we would consider it.

ejalaa12, 2022-03-17:
Sure, I can open it.

closed 2023-03-15 by clalancette

okalachev, 2023-03-16:
Why 0 is interpreted as variance unknown? Wouldn't it be better using `NAN` for that?

I mean, the variance can be equal to zero, if this is an absolute precise (ideal) data, like from a simulation.

ejalaa12, 2023-03-16:
That's the convention for all other messages. This is defined in ROS [Rep-145](https://www.ros.org/reps/rep-0145.html#id21)
Another why, I believe nan might not be a good idea is that It can break computation in other nodes, if they don't safe check for nans.

PR #181 (2023-03-15): sensor_msgs/Range lacks variance field
Fixes #180

As stated in the original issue, the Range msg lacks the variance field as opposed to most sensor msgs.

A similar issue exist for ros1, and here the discussion about it: https://github.com/ros/common_msgs/issues/142
commit 05d7e194 (2023-03-15): sensor_msgs/Range lacks variance field (#181)

* sensor_msgs/Range lacks variance field
Fixes #180