ros2/common_interfaces #188: Be able to support compressed point cloud messages (sensor_msgs/CompressedPointCloud?)

mlveggo, 2022-04-27:
## Feature request

#### Feature description
Lidar point clouds continue to grow and visualizing lidar point in realtime can use a lot of data. Practical amount using for example 4 spinning 128 channel lidars on a vehicle will grow above 1Gb/s.

So I would propose to support some sort of compressed point cloud (gzip:ed, quantization, mpeg-compressed) -> sensor_msgs/CompressedPointCloud.

Some compression ideas:
- GZip, Usually reduces point cloud data size with approximately 50%.
- https://mpeg-pcc.org/index.php/publications/an-overview-of-ongoing-point-cloud-compression-standardization-activities-video-based-v-pcc-and-geometry-based-g-pcc/
- https://cesium.com/blog/2019/02/26/draco-point-clouds/

#### Implementation considerations
Add a new standard sensor_msgs definition that can be used for compressed point clouds.

clalancette, 2022-04-27:
I think it is a good idea in general.  Before putting one in, I think it would be good to do research on the state-of-the-art for compressing point clouds.  Once we have that in hand, we would probably want to design a message that can compress to any of the available compression methods, so we'd want metadata in the beginning to describe the compression scheme and other necessary things.  Something like https://github.com/ros2/common_interfaces/blob/master/sensor_msgs/msg/CompressedImage.msg (but probably with more metadata fields).

gavanderhoorn, 2022-05-12:
Related discussion: [Compressed PointCloud2?](https://discourse.ros.org/t/compressed-pointcloud2/10616) on ROS Discourse.

Quite a few links to various existing implementations.