ros2/common_interfaces #201: visualization_msgs/msg/Marker missing text in TEXT_VIEW_FACING

delipl, 2022-08-03:
## Bug report

**Required Info:**

- Operating System:
  - OS: Arch Linux x86_64 
- Installation type:
  - Docker: `ros:humble`
- Version or commit hash:
  - `humble`
- DDS implementation:
  - `rmw_fastrtps_cpp`
  - `rmw_cyclonedds_cpp`
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

Cpp code:
```cpp
    heatpoints_marker_pub_ = create_publisher<visualization_msgs::msg::Marker>(HEATPOINTS_TOPIC_NAME, 10);
    visualization_msgs::msg::Marker heatpoints_min_value_marker_;
    heatpoints_min_value_marker_.header.stamp = now();
    heatpoints_min_value_marker_.header.frame_id = MAP_FRAME_NAME;
    heatpoints_min_value_marker_.type = visualization_msgs::msg::Marker::TEXT_VIEW_FACING;
    heatpoints_min_value_marker_.action = visualization_msgs::msg::Marker::ADD;
    heatpoints_min_value_marker_.scale.z = 1.0;
    heatpoints_min_value_marker_.pose.position.z = 0.2;
    heatpoints_min_value_marker_.ns = "values";
    heatpoints_min_value_marker_.color.a = 1.0;
    heatpoints_min_value_marker_.frame_locked = true;
    heatpoints_min_value_marker_.text = "should see text";
    heatpoints_min_value_marker_.mesh_resource = "should see text mesh";
    heatpoints_marker_pub_->publish(heatpoints_min_value_marker_);
```
Result 
`ros2 topic echo /heatpoints`
```yaml
header:
  stamp:
    sec: 1659528673
    nanosec: 138191064
  frame_id: map
ns: values
id: 0
type: 9
action: 0
pose:
  position:
    x: 0.0
    y: 0.0
    z: 0.2
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
scale:
  x: 0.0
  y: 0.0
  z: 1.0
color:
  r: 0.0
  g: 0.0
  b: 0.0
  a: 1.0
lifetime:
  sec: 0
  nanosec: 0
frame_locked: true
points: []
colors: []
text: '' # here empty
mesh_resource: '' # here empty
mesh_use_embedded_materials: false
---
```

#### Expected behavior
Attribute `text` and even `mesh_resource` should contain `string` values like a `ns` attribute.
#### Actual behavior
Attribute `text` and `mesh_resource` are empty but `ns` not.
#### Additional information
These are tries that don't work also:
```cpp
std::string text = "should see text";
heatpoints_min_value_marker_.text = text;
--------------------------------------------------------
std::string text = "should see text";
heatpoints_min_value_marker_.text = text.c_str();
```
I could not find any `visualization_msgs::msg::Marker::TEXT_VIEW_FACING` implementation in ros2 and I'm not sure if this example is even correct.

Best regards,
Jakub
----

clalancette, 2022-08-03:
For what it is worth, I can't reproduce this on Rolling on Ubuntu 22.04.  When I take your code above, I see the text values in the `text` and `mesh_resource` fields just fine.

Since Arch isn't one of our supported platforms, I'll suggest that you retry your example on Ubuntu 22.04 to see if you can reproduce there.

delipl, 2022-08-03:
Thanks for the quick response.
I understand that Arch is not supported but the node spins inside Docker container on `ros:humble` image.
Also tested on Ubuntu 22.04 in Docker image `ros:rolling` without result.

It looks like maybe docker problem.

delipl, 2022-08-03:
My listener container was on the `galactic` version. I didn't see this #153

closed 2022-08-03 by delipl