ros2/common_interfaces #203: Why is there no safety scanner standard msg?

tonynajjar, 2022-09-01:
I'm talking about the safety scanners that have fields of different shapes and output whether an obstacle is detected within that field.

What is your opinion on having a standard msg for that? I don't think any other existing "common" msg is suitable.

gbiggs, 2022-09-01:
I do not know why we don't currently have one, but you are most welcome to propose such a message! It would be helpful to do it as part of providing (or modifying) a safety scanner driver to publish the message, though, to ensure the design is driven by a use case and meets the needs of the hardware.

tonynajjar, 2022-09-01:
Thanks for the quick feedback. I'm not working on/with any scanner driver unfortunately and don't have much knowledge on the HW requirements. I was just hoping to use it.
I hope someone who does sees this issue and can take it on.