ros2/common_interfaces #207: Non-ASCII character causes an issue when parsing BatteryState.msg with 3rd party tools

zatevakhin, 2022-11-09:
Working with 3rd party tool to parse IDL messages get hit by UnicodeDecodeError caused by a non-ASCII symbol in [BatteryState.msg](https://github.com/ros2/common_interfaces/blob/rolling/sensor_msgs/msg/BatteryState.msg#L27) file.

[image]

See snippet below
```bash
...
# Power supply technology (chemistry) constants
uint8 POWER_SUPPLY_TECHNOLOGY_UNKNOWN = 0 # Unknown battery technology
uint8 POWER_SUPPLY_TECHNOLOGY_NIMH = 1    # Nickel–Metal Hydride battery
                                               ---^
uint8 POWER_SUPPLY_TECHNOLOGY_LION = 2    # Lithium-ion battery
...

```

clalancette, 2022-11-09:
Hm, that's true.  ASCII `-` should suffice here.  If you'd like to open a PR to fix it, we'd be happy to review.

That said, in the future we are going to allow non-ASCII in the message files, so it may be worthwhile to update your tools to deal with that.

zatevakhin, 2022-11-09:
Thanks! Created a PR https://github.com/ros2/common_interfaces/pull/208
> That said, in the future we are going to allow non-ASCII in the message files, so it may be worthwhile to update your tools to deal with that.

The parser was built using the ANTLR4 project. 
So maybe later, I'll create an issue for them about some Unicode symbols which seem not properly handled.

closed 2022-11-09 by gbiggs