ros2/common_interfaces #225: Image Projection Model

wodtko, 2023-07-28:
## Feature request

#### Feature description
Add a flag to the camera_info message which tell if an image should be rectified using a pinhole or spherical projection model.
When, e.g., using fisheye cameras, a pinhole projection model is not suitable. With FoVs of almost or even more than 180 degree, a different model is required.
This information is not available within the camera_info.

The distortion model must not be used for such information, since different projection models can be used for different kind of lenses/distortions.

#### Implementation considerations
For my specific use case, a simple flag (bool) would suffice to encode the required information.
I'm not sure though, if a  enum (int) might be the better option, since there are more possible ways of projecting images.
(pinhole, cylindrical, spherical, ...)

mjcarroll, 2023-07-28:
In theory, you can pack whatever distortion model you want between the string field and the distortion parameters.  It's not clear to me why that wouldn't be sufficient here?

wodtko, 2023-07-28:
Well, yes I can put what ever information into the distortion model string, sure.
But I would consider the distortion model describing lens distortions as something else than the projection model used during rectification.

Maybe a specific example might help here:

I implemented different distortion models for different kind of lenses and different approaches used for calibration
For this example this could be:
1. plumb_bob
2. fisheye_opencv
3. fisheye_matlab

now I put the respective name into the distortion model field.
Further I have different distortion parameter for all of these approaches and put them into the respective array.

For each of the undistortion approaches I can now decide whether the undistorted image should be created using a pinhole projection, a spherical projection, or a cylindrical projection.
The projection model used **is independent of the distortion model** used to describe the lense distortion.

When I add the information of the projection model to the distortion_model field I would end up having 9 different configurations/names. 
Possible yes, however, I would prefer both information in a separate field (e.g., a string or enum).