ros2/common_interfaces #275: Proposal - New nav_msgs/Trajectory message type

SteveMacenski, 2025-04-04:
### Description

In line with having sharp, pointed tools, I think we need a `nav_msgs/Trajectory` message to augment `nav_msgs/Path` to include time series information from modern controllers like MPC and Reinforcement Learning.

The `Path` only includes `geometry_msgs/PoseStamped` for each via point and we also need to be able to communicate velocity, acceleration, and possibly force/torque. 

### Motivation

See description above. 

### Design / Implementation Considerations

There already exists today the `trajectory_msgs` package, but these broadly apply to manipulators or multi-joint actuation systems. I don't believe that in the theme of conversations we've had recently on `Goals` vs `PoseStampedArray` that reusing that for a vasting different application is the maintainer's intent. 

I am creating temporary versions of these in `nav2_msgs` so we can keep moving forward, but given the dominance of MPC-based techniques (and RL, others)  that have time-varying trajectory outputs, I think this is an important general standard for ROS interfaces 

### Additional Information

I propose the following, which make full use of standard `geometry_msgs` fields for each data type:

`nav_msgs/Trajectory`

```
# An array of trajectory points that represents a trajectory for a robot to follow.

# Indicates the frame_id of the trajectory.
std_msgs/Header header

# Array of trajectory points to follow.
TrajectoryPoint[] points
```

`nav_msgs/TrajectoryPoint`

```
# Trajectory point state

# Desired time from the trajectory start to arrive at this trajectory sample.
builtin_interfaces/Duration time_from_start

# Pose of the trajectory sample.
geometry_msgs/Pose pose

# Velocity of the trajectory sample.
geometry_msgs/Twist velocity

# Acceleration of the trajectory (optional).
geometry_msgs/Accel acceleration

# Force/Torque to apply at trajectory sample (optional).
geometry_msgs/Wrench effort
```

SteveMacenski, 2025-04-04:
See https://github.com/ros-navigation/navigation2/pull/5057 for example use-case

SteveMacenski, 2025-09-02:
> My first thought is that we need to be encoding the expected reference frame into the message.
> The most complete of this would be a full Header with timestamp and frame_id

Meaning, `TrajectoryPoint`? It is in the header of `Trajectory`. I don't feel strongly, so sure! We could either use a `Header` , or add in a separate frame string. I think header would be nice, but it would need to be OK to semantically know that these aren't representing the _computation time_ of the Trajectory Point, but rather the timestamp in which that Point is representing. That's not usually how Header is used in my experience but I think it is a correct usage. 

Thoughts?

> If instead of storing the time position as a delta from another number it would actually make sense to store the value of T when the TrajectoryPoint is expected to be passed

Agreed, especially if we use a `Header`.

> But I think at least a comment would be important to pick how to define not-supplied.

I think this is the best course. There are several, especially sensor_msgs, that don't have explicit flags about its complete implementation, its just that `laser_msg.intensities.size() == 0` or `getYaw(imu_msg.orientation) == 0.0`. 

> And lastly if we're thinking about these as goal trajectories should we be providing space for tolerances for what's considered close enough etc. That may be the wrong level of abstraction for this message but I wanted to bring it up before we go forward without it.

Mhm, interesting. I agree that would be useful, but we also don't do that with nav_msgs or trajectory_msgs to date, so I'd probably air on the side of 'not' for consistency reasons alone. If constraints like that were introduced, I think there are many shapes they could take which would never quite fulfill everyone's needs. Though, I think in an announcement on Discourse for comment about this, that can be a point brought up for feedback. 

---

So what I'm taking away is to use a Header with a frame for each TrajectoryPoint (TrajectorySample?) + change the timestamp to be absolute and not relative. Also, add a comment about the nature of the acceleration and F/T if not populated. 

I am on board with that. Let me know what you think and if I'm OK to open up a draft PR with this.

Timple, 2025-09-13:
> Yeah, but since these fields are fixed size you can't check size. I'd suggest something like "If unused set twist.linear.x to NaN."

Would it make sense to make that the default then?
If people don't set it, it's automatically indicated as unused. Instead of accidentally communicated as zero.

SteveMacenski, 2025-09-16:
PR: https://github.com/ros2/common_interfaces/pull/296