ros2/common_interfaces #282: Add "Simulated" constelation for NavSatStatus

EmmanuelMess, 2025-05-14:
There is the possibility that the `NavSatStatus` message is being generated from Gazebo, in that case, there might be no "reference" constellation to set, so an option to set "simulated constellation" should be added, either as an extra boolean or as a bit for the `service` field.

peci1, 2025-05-14:
Isn't setting 0 enough? I always understood 0 (SERVICE_UNKNOWN) as "cannot tell". Or is there are any real benefit in knowing the the localization source is simulation?

Also, beware that there are physical GNSS simulators (i.e. wireless transmitters that behave similar to the constellation statellites). Should these also be covered?

EmmanuelMess, 2025-05-14:
> Isn't setting 0 enough? I always understood 0 (SERVICE_UNKNOWN) as "cannot tell". Or is there are any real benefit in knowing the the localization source is simulation?

In the interest of providing correct information in the message, the service is not "unknown", it is simulated. Which might in turn mean that the constellations at the time of the recording will not predict things like DOP.

> Also, beware that there are physical GNSS simulators (i.e. wireless transmitters that behave similar to the constellation statellites). Should these also be covered?

Ideally yes.