ros2/common_interfaces #289: GPS FLOAT case is missing

mattiadutto, 2025-07-01:
Hello to everyone, I'm working with UGV, and the next big step for my work is to add GPS to the system. 
In my facility, we are using EMLID Reach products, in particular RS+ as a base and M+ as a receiver. 
The GPS signal can be in 3 different states: SINGLE / FLOAT / FIX. Unfortunately, the current version of the [nmea_driver_sat](url) is managing in the same way as FLOAT and FIX. Looking at the [NavSatService](https://github.com/ros2/common_interfaces/blob/rolling/sensor_msgs/msg/NavSatStatus.msg) message, I discovered that it is missing the case for FLOAT messages.

What I propose is to add a FLOAT case in one of the two following ways:
```
int8 STATUS_UNKNOWN = -2        # status is not yet set
int8 STATUS_NO_FIX =  -1        # unable to fix position
int8 STATUS_FIX =      0        # unaugmented fix
int8 STATUS_SBAS_FIX = 1        # with satellite-based augmentation
int8 STATUS_GBAS_FIX = 2        # with ground-based augmentation
int8 STATUS_GBAS_FLOAT = 3  # with ground-based augmentation
```
Or in a more logical way
```
int8 STATUS_UNKNOWN = -2        # status is not yet set
int8 STATUS_NO_FIX =  -1        # unable to fix position
int8 STATUS_FIX =      0        # unaugmented fix
int8 STATUS_GBAS_FLOAT = 1 # with ground-based augmentation
int8 STATUS_SBAS_FIX = 2       # with satellite-based augmentation
int8 STATUS_GBAS_FIX = 3        # with ground-based augmentation
```

If this is something that can work, I can open a PR and later open another one to edit nmea_driver_sat

On nmea_driver_sat, there is also this [issue](https://github.com/ros-drivers/nmea_navsat_driver/issues/107) that is talking about the missing FLOAT case.

Ryanf55, 2025-07-12:
If we add enums, does that invalidate bags? Otherwise, I'm a fan of adding a float status. My Trimble PX-1 would use that (actually, it reports a PPP status, which is supported in MAVLink and GPSD, but not ROS 2 NavSatFix). 
Thanks for the ping! 
Yes, I'm always happy to work on interfaces, we have a seemingly larger amount of people interested in more descriptive but vendor-agnostic GPS messages.

peci1, 2025-07-17:
> So the idea to add this line `int8 STATUS_GBAS_FLOAT = 3 # with ground-based augmentation` is it possible?

Sounds good to me.

> do you have thoughts/opionions or desire to start a working group here?

I don't have the capacity to lead a WG, but if Ryan or someone else leads, I'll be glad to be a part of the WG.