ros2/common_interfaces #292: `geometry_msgs/PolygonIsntanceStamped` is missing in humble arm64

Ericsii, 2025-07-10:
I trying to build nav2 on an arm64 platform. Then I found `geometry_msgs/PolygonIsntanceStamped` is missing. In this #285 issue, I found you have release a new version. However, it seems didn't effect the arm64 platform.

Ericsii, 2025-07-10:
Here's my OS info: Ubuntu 22.04.5 LTS (GNU/Linux 6.8.0 aarch64)

christophebedard, 2025-07-17:
> > In the meantime, you can use the testing repo to get version 4.9.0, see https://docs.ros.org/en/rolling/Installation/Testing.html#deb-testing-repository.
> 
> I'll try this and keep an eye out for the next patch release. Thanks again!

Note that this page is outdated, so following the instructions will most likely not work. See https://github.com/ros2/ros2_documentation/issues/5773

claraberendsen, 2025-07-17:
> Note that this page is outdated
@Ericsii I opened a [PR](https://github.com/ros2/ros2_documentation/pull/5783) to update the docs for rolling if you want to follow the instructions there, or you can also follow the ones on the [kilted docs](https://docs.ros.org/en/kilted/Installation/Testing.html#deb-testing-repository) that should be the same minus this command in which the distribution name should be humble
```
sudo apt install ros-humble-my-just-released-package
```

christophebedard, 2025-07-22:
The sync is done, and [`geometry_msgs` for arm64 was definitely updated to 4.9.0](https://repo.ros2.org/status_page/ros_humble_ujv8.html?q=geometry_msgs), so I'll close this.

https://discourse.openrobotics.org/t/new-packages-and-patch-release-for-humble-hawksbill-2025-07-22/49221

closed 2025-07-22 by christophebedard